oreboot/oreboot #2: Turn on CI

rjoleary, 2019-05-24:
* Use azure pipelines
* Run tests in qemu

oxr463, 2019-06-28:
Any particular reason for using that instead of GitLab CI/CD, or even Travis?

rjoleary, 2019-06-29:
I'm not sure how the CI systems compare. There's also circleci which seems to be popular.

Any advice on how they compare would be super useful!

oxr463, 2019-06-29:
There are minor differences between the various offerings. I prefer buildbot, but it needs to be hosted, (similar to Jenkins). My second choice after that would be GitLab CI/CD. Both of these solutions are open source software! Then from there you have the various services, Travis, Circle, etc. If I had to choose, I'd prefer Travis to Circle simply because it's more widely used, at least for many of the projects that I've seen.

They all basically require a yaml configuration file included in the source, and some of them have better support for different platforms, languages, etc. For instance, Appveyor seems really popular for Windows.

Ideally, most of your test logic should be handled via `cargo test`; anything outside of that should be put in a script that can be run in a Docker container. Then, you could use multiple services without having to duplicate efforts for each.

rjoleary, 2019-07-13:
Thanks for the recommendations! I'll try setting up a very simple rustfmt with gitlab CI and see how that goes.

rjoleary, 2019-07-13:
See my comments on trying on gitlab ci in #25. Unfortunately, it did not quite work out.

oxr463, 2019-07-13:
That is unfortunate, but definitely understandable.