oreboot/oreboot #25: Check that rustfmt was run for each pull request

rjoleary, 2019-06-11:


oxr463, 2019-06-28:
```yaml
before_script:
- rustup component add clippy rustfmt
script:
- cargo fmt --all -- --check
- cargo clippy -- -D warnings
- cargo build
- cargo test
```

Taken from , [proot-me/proot-rs/.travis.yml](https://github.com/proot-me/proot-rs/blob/master/.travis.yml) (See: https://github.com/proot-me/proot-rs/issues/5).

rjoleary, 2019-07-13:
I'm not too sure what the issue is, but gitlab pipelines are only being run for merged pull-requests: https://gitlab.com/rjoleary/oreboot/pipelines

rjoleary, 2019-07-13:
It looks like gitlab CI works by having a mirror of the repository here: https://gitlab.com/rjoleary/oreboot. Github pushes events to gitlab on certain events to update the mirror. Updates to the mirror trigger pipelines to be run.

I enabled the "Push" and "Pull Request" events and can see these events being generated from the github settings dashboard. However, the issue is gitlab is not pulling the code until the PR has been merged to master.

Note the PRs are coming from forks, which might be why.

This is a bit of a deal breaker. I'll take another look through the settings sometime this weekend.

rjoleary, 2019-07-13:
I found some discussion here: https://gitlab.com/gitlab-org/gitlab-ee/issues/5667

It appears running gitlab ci on pull requests from forks in github is not supported. Some people suggested some additional bots might help, but it is iffy.

doriath, 2020-09-14:
#276 implemented this.

PR #56 (2019-07-13): Ci
Fixes #25
closed 2020-09-14 by doriath