oreboot/oreboot #65: Add compile-time check to guarantee a driver have exclusive access to its MMIO block

rjoleary, 2019-07-17:
If you initialize two drivers using the same MMIO register block, oreboot will still compile, but the drivers will conflict at runtime. For example:

```
// Notice these two drivers incorrectly reference the same MMIO register block.
let driver1 = &mut NS16550::new(0x1E78_3000, 115200);
let driver2 = &mut NS16550::new(0x1E78_3000, 9600);
driver1.init();
driver2.init(); // Second driver messes up the first driver's memory block.
```

Ideally, there should be some sort of compile-time check to ensure drivers have exclusive access to register blocks.

fotonick, 2020-11-20:
There is an idiom for this—see the [Singleton](https://rust-embedded.github.io/book/peripherals/singletons.html#how-do-we-do-this-in-rust) section from the Embedded Rust Book.

fotonick, 2020-11-23:
Ok, I implemented the SIngleton trait and it doesn't work on NS16550, the first Driver I tried.

Here's an example with a `Vec<usize>` that prevents taking multiple handles to the singleton payload as intended: [Rust Playground](https://play.rust-lang.org/?version=stable&mode=debug&edition=2018&gist=5afb83fff240ed99cd00fc25d315abf3). And here's the code I tried in `src/mainboards/emulation/qemu-riscv/src/main.rs`, which did not complain at all:

    struct NS16550Singleton {
      inner: Option<NS16550>,
    }

    impl NS16550Singleton {
      const fn new() -> Self {
        NS16550Singleton { inner: Some(NS16550::new(0x10_000_000, 115_200)) }
      }

      unsafe fn take(&mut self) -> NS16550 {
        let p = self.inner.take();
        assert!(self.inner.is_none());
        p.unwrap()  // Will panic if you try to take more than once.
      }

      unsafe fn restore(&mut self, t: NS16550) {
        let old_value = replace(&mut self.inner, Some(t));
        assert!(old_value.is_none());
      }
    }

    // Define a global singleton that you can't .take() twice.
    static mut UART: NS16550Singleton = NS16550Singleton::new();

    #[no_mangle]
    pub extern "C" fn _start(fdt_address: usize) -> ! {
        let mut uart0: NS16550 = unsafe { UART.take() };
        let mut uart1: NS16550 = unsafe { UART.take() };  // doesn't explode. =(
        ...
    }

When I examine the static `UART` after calling `.take()`, I see it still has its original value instead of `None`. I wonder if it's because it implements the `Deref` trait. I would next want to try putting it in a container like a tuple or newtype, but I am out of time to work on oreboot for the week, so here are my scribbles for the next person who wants to take this up. (Maybe me.)

orangecms, 2022-04-30:
I have seen this working in https://github.com/esp-rs 's HAL crate.
We should check there how they solved it.