oreboot/oreboot #67: Setup system to allow end-users to configure the build

rjoleary, 2019-07-17:
Coreboot has kconfig and "make menuconfig". This provides UI to discover and change build settings.

Right now for oreboot, we use [conditional compilation](https://doc.rust-lang.org/1.30.0/book/first-edition/conditional-compilation.html) to select which drivers to include. This does not seem to extend far beyond turning features on/off.

For example, we need a mechanism for the end-user to specify a path to the kernel they want to include.

rjoleary, 2019-09-08:
@sphinxc0re I set you as the assignee since you had a bunch of ideas on this topic.

pickfire, 2020-02-11:
Maybe we can combine conditional compilation with environment variables which allow us to set the paths for some particular configuration?

But I wonder if conditional compilation could have constrains embedded in it like how coreboot does. Like determining the hierarchy of configuration.

orangecms, 2023-03-11:
We have started using xtask as a build framework, passing on build options to e.g. opt into SBI for sunxi/nezha (to be generalized for RISC-V) as well aa the payload.

orangecms, 2024-01-16:
Having come a long way, we now offer the first build options, including DRAM size for the JH7110 (StarFive VisionFive 2, Milk-V Mars, ...).

orangecms, 2024-01-16:
For thr DRAM size, I included build-time checks. It errors when provided unsupported values.

Closing this as complete now.

closed 2024-01-16 by orangecms