oreboot/oreboot #70: Coroutines

rjoleary, 2019-07-19:
- Polling I/O is very slow
  - A few UART prints = 0.01s
  - Read from SPI and verify loop
- Interrupt-based I/O is difficult to do well
  - Puts us on slippery slope to becoming a kernel
  - Non-preemptive threading has been shown to be “good enough” in firmware
- Implementation details
  - Save state
  - Long jump
  - (simple) Scheduler -- round robin has been shown to be good enough

fotonick, 2020-05-22:
Could we use the new [no_std async/await](https://ferrous-systems.com/blog/stable-async-on-embedded/)? I think it's way easier than writing a scheduler.

fotonick, 2020-11-20:
Ryan, can you point to a specific file / subsystem that could be improved with concurrency? The simpler first example, the better.

orangecms, 2024-01-16:
We are using coroutines (formerly generators, renamed in Rust a few months ago) in the SBI for RISC-V.