oreboot/oreboot #72: Design vboot system for oreboot

rjoleary, 2019-07-19:


rjoleary, 2021-06-26:
We should probably just borrow coreboot's vboot architecture. The first task would be to research coreboot's vboot.

jhand2, 2021-10-04:
If this is something we're still interested in, I would be happy to look into this once we've settled some of the FSP stuff on upxtreme.

derpsteb, 2022-01-27:
So I read a bit about this topic and talked to Jordan if he wants some help on this. 
My main resource here were the design docs on chromium.org. 

In general we would like to have a solution that allows owners of a device to claim full ownership. This prohibits the use of fused keys. Therefore I propose the following (based on [ChromeOS RoT](https://www.chromium.org/chromium-os/chromiumos-design-docs/verified-boot-data-structures/#root-of-trust) and what others have mentioned on slack):

- [ ] **root key**: Add fdt node that specifies location, size and other relevant metadata. points to read-only flash. used to verify the firmware keyblock.
- [ ] **firmware keyblock**: Add  fdt node with location, size and other metadata. located in read-write flash. contains key used to verify the payload.
- [ ] **area.rs**: Add code to parse the two new node types.
- [ ] **layoutflash**: Add code to differentiate between ro and rw flash. Add code to place the firmware keyblock in the respective space.
- [ ] **vboot driver**: Add code to load root key, verify firmware keyblock, unpack keyblock and verify payload.

Note: I am not sure about the separation into ro/rw since my current understanding is that oreboot lives entirely in ro flash. Does it make sense to introduce this distinction? 

I think having a measured boot implementation as well would also be nice, but this is a separate topic.