oreboot/oreboot #100: Use caching in azure pipelines to speed up CI

rjoleary, 2019-08-11:


fotonick, 2020-04-25:
Consider a shallow git clone (`git clone -–depth 1`) to speed that phase up.

Azure is pouring a ton of time into building QEMU 4.2. And are you willing to point Debian to the testing package repo? I see that [4.2 is available on testing](https://packages.debian.org/search?keywords=qemu).

mpictor, 2020-09-20:
If people are OK with using a pre-built binary, one solution I've used (that doesn't involve storage in git or use of a paid blobstore) is to upload a tarball to github as an asset for some particular release, such as the very first. Then, just download the tarball on demand.

I've done that with a custom compiler tarball that bazel [downloaded](https://github.com/mpictor/bazel-amd64-crosstool/blob/e6c5e492b3ac2f8f521ac316f731782e38eb9cbd/tools/deps.bzl#L9). Note that I always downloaded from the same release; to download from the latest, you'd have to upload again with each new release.

fotonick, 2020-09-21:
I spent some time looking at related ideas in PR #255. Our Azure CI provides artifact save and restore features. I found that caching the firmware binaries saves surprisingly little time because the Rust toolchain takes so much time to set up. (Rust toolchain is needed for both the build and test phases as we've set it up today; could be removed from test phase with some work, I think.) Caching the Rust toolchain itself saves surprisingly little time because of way-slower-than-I-expected network speeds to the cache (~10 MBps, IIRC).

I somehow didn't look into caching the QEMU build. I really should have, since it's ~4 minutes of the CI build, IIRC, and probably not such a huge binary size. That's a good idea.

mpictor, 2020-09-22:
If qemu is being built for a single well-defined target, you can also speed the build by disabling unneeded things. IIRC [this](https://github.com/purecloudlabs/gprovision/blob/master/build/qemu/Dockerfile#L87) made quite a difference.

mpictor, 2020-09-25:
If I change the qemu build step to this
```bash
    set -e
    git clone -b v5.1.0 https://github.com/qemu/qemu --depth 1 && pushd qemu
    mkdir build && cd build
    ../configure --target-list=riscv64-softmmu,x86_64-softmmu --audio-drv-list= --python=/usr/bin/python3 --disable-bzip2 --disable-curl --disable-curses --disable-docs --disable-gcrypt --disable-gnutls --disable-gtk --disable-guest-agent --disable-kvm --disable-libssh --disable-libusb --disable-libxml2 --disable-live-block-migration --disable-lzo --disable-mpath --disable-nettle --disable-numa --disable-opengl --disable-rdma --disable-replication --disable-sdl --disable-snappy --disable-spice --disable-tpm --disable-usb-redir --disable-vhost-crypto --disable-vhost-net --disable-vhost-scsi --disable-vhost-user --disable-vhost-vsock --disable-virglrenderer --disable-vnc --disable-vnc-jpeg --disable-vnc-png --disable-vnc-sasl --disable-xen --disable-debug-mutex --disable-debug-info --disable-debug-tcg --enable-virtfs
    make -j$(($(nproc) + 1)) qemu-img
    make -j$(($(nproc) + 1)) -C x86_64-softmmu qemu-system-x86_64
    make -j$(($(nproc) + 1)) -C riscv64-softmmu qemu-system-riscv64
    #make -j16
    popd
    mkdir bin
    ln -s $(realpath qemu/build/x86_64-softmmu/qemu-system-x86_64) bin/
    ln -s $(realpath qemu/build/riscv64-softmmu/qemu-system-riscv64) bin/
    ln -s $(realpath qemu/build/qemu-img) bin/
```
it takes ~2min on an _old_ 8-core box, not counting the checkout step.

I tried to configure with `--static` but hit an error I didn't feel like investigating.
```
ERROR: sizeof(size_t) doesn't match GLIB_SIZEOF_SIZE_T.
       You probably need to set PKG_CONFIG_LIBDIR
       to point to the right pkg-config files for your
       build target
```

mpictor, 2020-09-25:
Both risc-v tests produced the expected tests with this stripped-down qemu.

Without an x86 payload, I gather that it is to be expected that `make run` in emulation/qemu-q35 produces an error like
```
Welcome to oreboot
loading payload with fdt_address 0
loading ...
decode header at 0
decode header now at 28
PANIC: panicked at 'Panic'ing on PAYLOAD_SEGMENT_BAD: seg now 4278190080 CBFSSeg { typ: 4294967295, comp: 4294967295, off: 4294967295, load: 18446744073709551615, len: 4294967295, memlen: 4294967295 } typ PAYLOAD_SEGMENT_BAD', /home/mark/experiments/oreboot/payloads/src/payload/payload.rs:215:21
```

orangecms, 2022-02-05:
Note: We're using GitHub Actions now. It's spending quite some time to install our dependencies. Maybe that can be sped up.

The other thing we could cache is the Rust crates in `~/.cargo`. I think there is some GitHub Action to do that easily.

orangecms, 2022-04-30:
With the move to Cargo workspace and restructuring, we're up to speed again. Should we close this issue? Thoughts?

closed 2026-02-15 by orangecms