oreboot/oreboot #159: Add function to perform debug prints

rjoleary, 2019-09-08:
Currently, it's difficult to perform "printf() debugging". If you want to place a debug print into an arbitrary function, you must pass the `uart0` driver or `w` through the entire call stack which can be a fair bit of work.

It would be much more convenient if there were some global debug print function.

However, it seems Rust is averse to having global mutable state like this. From my current understanding, there would need to be an `unsafe` somewhere in there (which I think is okay if there's no other way).

alistair23, 2019-09-09:
From my understanding as well the only way to have a global mutable is with an unsafe. In this case I think it's justifiable as we have already seen this is important for the panic handler. I suspect there will be more and more cases where this will be useful.

andre-richter, 2019-09-09:
Youll eventually need an unsafe, yes.

In case you plan to use the printing from multiple cores at some later point in time, you can think of proactively adding a locking interface already for which you can exchange the implementations to your need as you go. If not, don't read on 😉

My experience with the issues you just encountered are written down at:
https://github.com/rust-embedded/rust-raspi3-OS-tutorials/blob/master/0F_globals_synchronization_println/README.md

Maybe it can provide a useful hint or two.

derpsteb, 2021-12-22:
It might be useful to use the `cfg` attribute in order to only include the debug print() during `debug` mode builds[1].
Debug assertions might be the more compact way to do this in most cases.
The `cfg!` macro can also be used to write if-statements.

[1] https://doc.rust-lang.org/reference/conditional-compilation.html#the-cfg-attribute
[2] https://doc.rust-lang.org/reference/conditional-compilation.html#debug_assertions

orangecms, 2024-01-16:
We have rich printing macros by now, with some extra small utilities for when even the UART isn't working well, or when linking at wrong addresses etc..

Closing this as complete. :-)

closed 2024-01-16 by orangecms