oreboot/oreboot #160: Spin harts in the bootblob, then the romstage and later the payloader stage

rjoleary, 2019-09-08:
I believe this was the suggestion from @philhug . 

1. Initially, the harts (other than the boot hart) spin in the bootblob.
2. Once the romstage is loaded, they are transferred to spin in the romstage.
3. Once the payloader stage is loaded, they are transferred to spin in the payloader stage.

Currently, cores are spinning in the bootblob.

orangecms, 2023-02-11:
I am doing that in https://github.com/oreboot/oreboot/pull/606

As it turns out, often enough do we not need more than a single stage, and the mechanism for resuming harts would be per SoC AFAICT. Any further thoughts?

orangecms, 2024-01-16:
Sooo... as it turns out, for OSes relying on an SBI (i.e. those targeting S-mode), we need a second stage indeed. So we keep spinning on the JH7110, as of now. And yes, it is per SoC / platform.

There is a WIP PR #694 for the SBI on JH7110, which I got to run Linux on a sigle hart so far.