oreboot/oreboot #161: Adapt oreboot runtime based on DT

alistair23, 2019-09-09:
As mentioned in https://github.com/oreboot/oreboot/pull/144 oreboot could look at the device tree that has been passed to it and use that to decide what to initialise. The DT could also be used to make the Rust code more generic as we could obtain addresses and other information from the DT instead of hard coding it.

This would probably increase the complexity of oreboot, but as we already had FDT logic it shouldn't be too bad.

Thoughts?

rminnich, 2019-09-09:
you want to be a bit careful here. Vendors screw these things up all the time. So for things such as, e.g., the ddr controller, I'm more comfortable with hardcodes. you can't afford for that to go wrong. There might be other things, involving options, for which the DT makes sense. 

So maybe for the early going, i.e. DRAM and clock enabling, we take non-authoritative hints.

But there are few things worse than getting a new batch of chips and having them all go dark on you.

alistair23, 2019-09-09:
What would a vendor screw up?

We would either use the Linux kernel DT or maintain our own DTs (like U-Boot).

alistair23, 2020-02-05:
I'm going to close this as it seems like this isn't the agreed upon approach.

closed 2020-02-05 by alistair23