oreboot/oreboot #162: Checksum payloads

rjoleary, 2019-09-10:
The checksum would check the integrity of the payload and print an error if it doesn't match. This isn't for security.

fotonick, 2020-09-18:
At what stage are you proposing to checksum? At what stage are you proposing to enforce?

rjoleary, 2020-11-25:
I imagine it would be in the payloader stage. You can see how coreboot does it here: https://github.com/coreboot/coreboot/blob/master/Documentation/cbfs.txt#L211

fotonick, 2021-02-05:
Here's my current understanding of what needs to be done:

1. At build time, insert a header before the bytes of the main payload. This would be in `layoutflash`.
2. At run time, verify the checksum against header bytes. This would be in `payload.rs`.

Does that sound right?

rjoleary, 2021-06-26:
Yes, that sounds right. A simple CRC checksum would do (ex: https://crates.io/crates/crc). Nothing cryptographic is needed.