oreboot/oreboot #183: Build is broken

alistair23, 2019-11-08:
oreboot is unable to link. I see this error:

```
error: linking with `rust-lld` failed: exit code: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "-L" "/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/sysroot/lib/rustlib/riscv64imac-unknown-none-elf/lib" "/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps/hifive-5d67b9f37960cb79.hifive.eefz7yfk-cgu.5.rcgu.o" "-o" "/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps/hifive-5d67b9f37960cb79" "--gc-sections" "-L" "/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps" "-L" "/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/release/deps" "-L" "/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/sysroot/lib/rustlib/riscv64imac-unknown-none-elf/lib" "-Bstatic" "/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/sysroot/lib/rustlib/riscv64imac-unknown-none-elf/lib/libcompiler_builtins-94979fc64732be66.rlib" "-Tlink.ld" "-Bdynamic"
  = note: rust-lld: error: undefined symbol: __atomic_load_16
          >>> referenced by compiler_builtins.4xlgpysu-cgu.3
          >>>               compiler_builtins-94979fc64732be66.compiler_builtins.4xlgpysu-cgu.3.rcgu.o:(__llvm_memcpy_element_unordered_atomic_16) in archive /home/vsts/work/1/s/src/mainboard/sifive/hifive/target/sysroot/lib/rustlib/riscv64imac-unknown-none-elf/lib/libcompiler_builtins-94979fc64732be66.rlib
          >>> referenced by compiler_builtins.4xlgpysu-cgu.3
          >>>               compiler_builtins-94979fc64732be66.compiler_builtins.4xlgpysu-cgu.3.rcgu.o:(__llvm_memmove_element_unordered_atomic_16) in archive /home/vsts/work/1/s/src/mainboard/sifive/hifive/target/sysroot/lib/rustlib/riscv64imac-unknown-none-elf/lib/libcompiler_builtins-94979fc64732be66.rlib
          >>> referenced by compiler_builtins.4xlgpysu-cgu.3
          >>>               compiler_builtins-94979fc64732be66.compiler_builtins.4xlgpysu-cgu.3.rcgu.o:(__llvm_memmove_element_unordered_atomic_16) in archive /home/vsts/work/1/s/src/mainboard/sifive/hifive/target/sysroot/lib/rustlib/riscv64imac-unknown-none-elf/lib/libcompiler_builtins-94979fc64732be66.rlib
          
          rust-lld: error: undefined symbol: __atomic_store_16
          >>> referenced by compiler_builtins.4xlgpysu-cgu.3
          >>>               compiler_builtins-94979fc64732be66.compiler_builtins.4xlgpysu-cgu.3.rcgu.o:(__llvm_memcpy_element_unordered_atomic_16) in archive /home/vsts/work/1/s/src/mainboard/sifive/hifive/target/sysroot/lib/rustlib/riscv64imac-unknown-none-elf/lib/libcompiler_builtins-94979fc64732be66.rlib
          >>> referenced by compiler_builtins.4xlgpysu-cgu.3
          >>>               compiler_builtins-94979fc64732be66.compiler_builtins.4xlgpysu-cgu.3.rcgu.o:(__llvm_memmove_element_unordered_atomic_16) in archive /home/vsts/work/1/s/src/mainboard/sifive/hifive/target/sysroot/lib/rustlib/riscv64imac-unknown-none-elf/lib/libcompiler_builtins-94979fc64732be66.rlib
          >>> referenced by compiler_builtins.4xlgpysu-cgu.3
          >>>               compiler_builtins-94979fc64732be66.compiler_builtins.4xlgpysu-cgu.3.rcgu.o:(__llvm_memmove_element_unordered_atomic_16) in archive /home/vsts/work/1/s/src/mainboard/sifive/hifive/target/sysroot/lib/rustlib/riscv64imac-unknown-none-elf/lib/libcompiler_builtins-94979fc64732be66.rlib
          >>> referenced by compiler_builtins.4xlgpysu-cgu.3
          >>>               compiler_builtins-94979fc64732be66.compiler_builtins.4xlgpysu-cgu.3.rcgu.o:(__llvm_memset_element_unordered_atomic_16) in archive /home/vsts/work/1/s/src/mainboard/sifive/hifive/target/sysroot/lib/rustlib/riscv64imac-unknown-none-elf/lib/libcompiler_builtins-94979fc64732be66.rlib
```

It happens locally and on Azure: https://dev.azure.com/azure0427/Oreboot%20Pipeline/_build/results?buildId=338

alistair23, 2019-11-08:
I have tried older toolchains and still see this failure. I'm not sure where it is coming from

rminnich, 2019-11-09:
We need to look for uses of u16.

The error message sucks, bigly, since we have no idea what part of oreboot called it :-(

rminnich, 2019-11-09:
it may be in serialize.rs

rminnich, 2019-11-09:
or not ...

alistair23, 2019-11-09:
Yeah, it's rough

asb, 2019-11-12:
CC @lenary 
In the GCC world this symbol is provided by libatomic rather than libgcc directly. In the LLVM world, compiler-rt provides the symbol. Rust should either use a RISC-V compiler-rt or if using libgcc it should link libatomic as well.

alistair23, 2019-11-12:
Closing the issue as this has been fixed.

closed 2019-11-12 by alistair23

lenary, 2019-11-13:
Yep.

The 'A' extension only includes instructions for 32- and 64-bit atomic accesses. If you want to do a smaller-than-32-bit atomic access, you need to link to some atomic access routines.

There are two different sets of atomic access routines available. There is libgcc's libatomic implementation, for which you need to pass `-latomic` when linking (because libatomic is not part of libgcc). If you're using compiler-rt, you don't need to link extra libraries as the atomic access routines are provided within the compiler builtins library.