oreboot/oreboot #217: oreboot needs flash header to load properly

bcantrill, 2020-01-12:
In Tock on OpenTitan, we discovered that Tock was broken by https://github.com/lowRISC/opentitan/commit/4425b287f8d2c6de6f10410192ee7ef761c71189, as described in https://github.com/tock/tock/pull/1460#issuecomment-573380139.  It appears that oreboot has the same problem:  its layout will need to be updated to reflect the flash header that the OpenTitan boot ROM is looking for.  To get oreboot working after https://github.com/lowRISC/opentitan/commit/4425b287f8d2c6de6f10410192ee7ef761c71189, the flash header will need to be at 0x20000000; this fix gets oreboot working again:

```diff
diff --git a/src/mainboard/opentitan/crb/link.ld b/src/mainboard/opentitan/crb/link.ld
index 712cec6cc3..691e922eef 100644
--- a/src/mainboard/opentitan/crb/link.ld
+++ b/src/mainboard/opentitan/crb/link.ld
@@ -28,6 +28,10 @@ SECTIONS
 {
 
     . = 0x20000000;
+    .flash_header :
+    {
+        LONG(_boot);
+    }
     .bootblock :
     {
         KEEP(*(.bootblock.boot));
```

Happy to open a pull request on this, if desired.  Though do note that integrating this fix will mean that oreboot will now fail to function on any OpenTitan that pre-dates https://github.com/lowRISC/opentitan/commit/4425b287f8d2c6de6f10410192ee7ef761c71189.  That there is no versioning in the flash header makes it too easy for OpenTitan payloads to break; an OpenTitan issue will be raised on that.

rminnich, 2020-01-13:
I think ian fixed that friday, can you double check. 

Delighted to see you looking at the code!!

bcantrill, 2020-01-13:
Ian's fix allows for Oreboot to run on OpenTitan that pre-dates https://github.com/lowRISC/opentitan/commit/4425b287f8d2c6de6f10410192ee7ef761c71189 -- but for more recent OpenTitan, you need the flash header (I verified that Oreboot was broken by the change -- and fixed by my suggested fix).  Unfortunately, there is no (easy) way to have a single fix that will work on OpenTitan before and after this change, but a flash header specification sounds like it's coming soon, so hopefully we won't have to endure more silent breakage of this manner!

iangoegebuer, 2020-01-14:
You are correct that the committed fix is for the older version of OpenTitan. Currently the readme references the Build Artifacts that contain the old version of the ROM code. When a new release comes out with a change to the linker is when it makes sense to update our linker. Otherwise we keep having to change our linker to match a moving target [https://github.com/lowRISC/opentitan/releases](https://github.com/lowRISC/opentitan/releases)

Ideally we will be placing our own ROM code in the bitstream and not have to rely on the one baked into the OpenTitan release

hugelgupf, 2020-01-14:
Are they even still doing those releases? Last one in November seems not too recent. I think moving target is the nature of this work at the moment, and to keep up you just have to keep moving with it.