oreboot/oreboot #218: Cargo-make issues

rjoleary, 2020-01-16:
cargo-make is deficient in the following areas:

1. **Config variables**: Ideally, the end user can customize compile-time constants such as the baudrate. Something like, but not necessarily, kconfig is desired.
2. **Duplication**: Each mainboard has a Makefile.toml which is nearly identical. Ideally, some of the logic here can be shared across mainboards.
3. **Incremental builds**: Changes to some files do not trigger a rebuild. A few times, I have accidentally used an older build.

alistair23, 2020-01-16:
meson supports Rust, maybe worth swapping?

rminnich, 2020-01-16:
it's really hard as hell to get it to do an offline build. Yeah, maybe meson. OpenTitan uses it.

gkelly, 2020-01-16:
We've had some inflexibility issues with meson. @mcy could speak more to it, but I don't know if I'd take our use of it as an outright endorsement.

sagiegurari, 2020-02-26:
actually cargo-make has options for all 3

1. config variables - you can use 'functions' in command args and swap the values. you can pass env vars in cli and you can setup env via duckscript. there are more options.
2. duplication - makefile.toml can extend each other. you can import tasks and so on from multiple other makefiles. fact that in this repo there are duplicates, doesn't mean it is a limitation in cargo-make
3. didn't fully understand this one, but cargo-make has built in watch support which should trigger right build command per area you changed.

i wouldn't assume that specific implementation of it in this repo defines what it can or cannot do.

fotonick, 2020-05-21:
Propose to close, given that we moved to regular Makefiles.

doriath, 2020-08-04:
+1 for closing it.

closed 2020-08-04 by fotonick