oreboot/oreboot #250: qemu-armv7 build broken

niting, 2020-04-28:
I am new to oreboot and was playing around with the armv7 build. It looks like the compilation of the armv7 build is broken. 

Steps to reproduce:

1. git clone git://github.com/oreboot/oreboot
2. cd oreboot
3. make firsttime
4. make update
5. cd src/mainboard/emulation/qemu-armv7
6. make

This leads to the following errors:
error[E0432]: unresolved import `payloads::external::zimage::PAYLOAD`
 --> src/romstage.rs:2:5
  |
2 | use payloads::external::zimage::PAYLOAD;
  |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ no `PAYLOAD` in `external::zimage`

error[E0432]: unresolved import `payloads::external::zimage::DTB`
  --> src/main.rs:13:5
   |
13 | use payloads::external::zimage::DTB;
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ no `DTB` in `external::zimage`

error[E0599]: no method named `write_fmt` found for struct `print::WriteTo<'_>` in the current scope
  --> src/main.rs:42:5
   |
42 |     write!(w, "{}{}\r\n", 3, "7").expect("blame ryan");
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ method not found in `print::WriteTo<'_>`
   |
   = help: items from traits can only be used if the trait is in scope
   = note: this error originates in a macro (in Nightly builds, run with -Z macro-backtrace for more info)
help: the following trait is implemented but not in scope; perhaps add a `use` for it:
   |
9  | use core::fmt::Write;

I suspect that the unresolved imports are due to a recent change that deleted unused variables. More than happy to fix this myself in a PR to get my hands dirty with the source code but wanted to confirm that this is indeed broken.

fotonick, 2020-04-28:
I can confirm that it's broken for several reasons. Most are very easy to fix (`asm` macro (#242), missing `Write` trait, and a missing `#![allow(non_snake_case)]`), but it looks like the `Payload` struct has changed substantially since this code last built, so it will take a little bit of study.

I'll note that one reason that `qemu-armv7` broke is because it's not covered by the CI. Once we fix it, let's see if we have the CI time budget to add it.

niting, 2020-04-29:
Agreed that the fixes not related to the Payload struct are simple. I will keep this one open for the Payload struct and see if I can work on it.

closed 2020-05-22 by fotonick