oreboot/oreboot #259: Build broken

fotonick, 2020-05-19:
Azure Pipeline indicates that the build broke on May 12. There was no push between May 11 and May 12, so it was almost certainly an upstream toolchain breakage.

What I have tried:
* I verified breakage locally by rerunning `make firsttime`. I note that the only package to actually update was `cargo-xbuild` from `v0.5.29` to `v0.5.32`. We have not so far been locking its version.
* Deleting `~/.cargo` and rerunning `make firsttime` again, I get the same issue.
* I note that the failure message changed between May 12 and today (May 19).
  * [Today](https://dev.azure.com/azure0427/Oreboot%20Pipeline/_build/results?buildId=843&view=logs&j=12f1170f-54f2-53f3-20dd-22fc7dff55f9&t=5caf77c8-9b10-50ef-b5c7-ca89c63e1c86&l=41), the first failure is:

        "/usr/local/google/home/foton/.rustup/toolchains/nightly-2020-04-23-x86_64-unknown-linux-gnu/bin/cargo" "rustc" "-p" "alloc" "--release" "--manifest-path" "/tmp/cargo-xbuild.WIcwcV20mj59/Cargo.toml" "--target" "riscv64imac-unknown-none-elf" "-v" "--" "-Z" "force-unstable-if-unmarked"
        error: failed to run `rustc` to learn about target-specific information

     When I run this command manually, I see the error "error: package ID specification `alloc` matched no packages".
  * On [May 12](https://dev.azure.com/azure0427/Oreboot%20Pipeline/_build/results?buildId=834&view=logs&j=12f1170f-54f2-53f3-20dd-22fc7dff55f9&t=5caf77c8-9b10-50ef-b5c7-ca89c63e1c86), it was:

        Compiling device_tree v0.1.0 (/home/vsts/work/1/s/src/lib/device_tree)
        Running `rustc --crate-name device_tree --edition=2018 /home/vsts/work/1/s/src/lib/device_tree/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type lib --emit=dep-info,metadata,link -C opt-level=z -C metadata=c1c7a5014ae86a46 -C extra-filename=-c1c7a5014ae86a46 --out-dir /home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps --target riscv64imac-unknown-none-elf -L dependency=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps -L dependency=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/release/deps --extern byteorder=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps/libbyteorder-ce606ccc0d608c41.rmeta --extern model=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps/libmodel-f62c0edc0a89d972.rmeta --extern num_derive=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/release/deps/libnum_derive-7ebed80a58639909.so --extern num_traits=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps/libnum_traits-970056b6746faf0a.rmeta --extern wrappers=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps/libwrappers-1e048120b0684a3f.rmeta -C link-arg=-Tlink.ld --sysroot /home/vsts/work/1/s/src/mainboard/sifive/hifive/target/sysroot`
        Running `rustc --crate-name hifive --edition=2018 src/main.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -C opt-level=z -C lto -C metadata=4833c18518e6e609 -C extra-filename=-4833c18518e6e609 --out-dir /home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps --target riscv64imac-unknown-none-elf -L dependency=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps -L dependency=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/release/deps --extern arch=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps/libarch-7b45bd4b11fcd81c.rlib --extern clock=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps/libclock-61ecc3456473c2ba.rlib --extern console=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps/libconsole-3f0b28e05cf15616.rlib --extern device_tree=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps/libdevice_tree-c1c7a5014ae86a46.rlib --extern heapless=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps/libheapless-920617368a039fd1.rlib --extern model=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps/libmodel-f62c0edc0a89d972.rlib --extern payloads=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps/libpayloads-f518dad430203026.rlib --extern print=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps/libprint-07309844fa1956d7.rlib --extern register=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps/libregister-781f558611a63a3c.rlib --extern soc=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps/libsoc-d7c48d4e949f531e.rlib --extern spi=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps/libspi-7021cd79dce821bd.rlib --extern static_ref=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps/libstatic_ref-ee067a35d5a2ad1e.rlib --extern uart=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps/libuart-29d5f4bc26c4b1e6.rlib --extern wrappers=/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps/libwrappers-1e048120b0684a3f.rlib -C link-arg=-Tlink.ld --sysroot /home/vsts/work/1/s/src/mainboard/sifive/hifive/target/sysroot`
        error: linking with `rust-lld` failed: exit code: 1
          |
          = note: "rust-lld" "-flavor" "gnu" "-L" 
        "/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/sysroot/lib/rustlib/riscv64imac-unknown-none-elf/lib" "/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps/hifive-4833c18518e6e609.hifive.8pjontx3-cgu.2.rcgu.o" "-o" "/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps/hifive-4833c18518e6e609" "--gc-sections" "-L" "/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/riscv64imac-unknown-none-elf/release/deps" "-L" "/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/release/deps" "-L" "/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/sysroot/lib/rustlib/riscv64imac-unknown-none-elf/lib" "-Bstatic" 
        "/home/vsts/work/1/s/src/mainboard/sifive/hifive/target/sysroot/lib/rustlib/riscv64imac-unknown-none-elf/lib/libcompiler_builtins-6a4465d8cbab49a7.rlib" "-Tlink.ld" "-Bdynamic"
        = note: rust-lld: error: undefined symbol: __atomic_load_1
        ... Many more undefined symbols

I have not yet finished going through the [cargo-xbuild issues list](https://github.com/rust-osdev/cargo-xbuild/issues) to figure out what caused this, what changed the error, or if the current issue has been reported.

fotonick, 2020-05-19:
Quick observations:
* In today's failure, the target triple is mismatched between what toolchain is being invoked (`x86_64-unknown-linux-gnu`) and the target (`riscv64imac-unknown-none-elf`). 
* [rust-osdev/cargo-xbuild#74](https://github.com/rust-osdev/cargo-xbuild/issues/74) is almost certainly related.

rjoleary, 2020-05-20:
"undefined symbol: __atomic_load_1". This reminds me of this hack: https://github.com/oreboot/oreboot/commit/ceba28d031fbb5e8e3e33dce3ca6d792ad95c311 (this has since been fixed). For some reason, the compiler tries generating unsupported atomic intrinsic functions.

Atomics is supported for **riscv64imac**. "a" in "imac" stands for "atomic". However, there is a limitation. According to the *The RISC-V Reader*, the atomic instructions either operate on W (4 bytes) or D (8 bytes). "__atomic_load_1" indicates this is an atomic load of a single byte. A native instruction does not exist for a single byte atomic load on riscv64imac.

You can probably simulate it by aligning the address to a multiple of 4 and using the `lr.w` instruction. However, I am not sure why the compiler is generating __atomic_load_1 to begin with.

fotonick, 2020-05-20:
Thanks for the connection, Ryan! However, the atomics thing was from the May 12 error, which no longer occurs. My guess is that he current error is from trying to compile the riscv target with an x86_64 toolchain and being unable to find `alloc`. Tomorrow, I'll try to work through why cargo-xbuild decided to do that. All this said, it's possible that the atomics problem is still there lurking underneath and we'll need to look back at the hack you pointed out.

PR #260 (2020-05-21): Pin cargo package versions
Pin cargo-xbuild and cargo-binutils versions in `make firsttime`. In doing so, revert cargo-xbuild by three versions (0.5.32 &rarr; 0.5.29), which fixes #259. I verified that all newer versions are broken for us, but have not yet taken the time to figure out whether the five issues reported against cargo-xbuild in the relevant time period correspond to our breakage or not.
closed 2020-05-21 by fotonick