oreboot/oreboot #283: Build error when building all mainboards

GanShun, 2020-09-18:
One of the boards is throwing up this error when I build all mainboards:

error[E0658]: `if` is not allowed in a `const`
  --> /home/ganshun/.cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.36/src/int/specialized_div_rem/mod.rs:68:22
   |
68 |   const USE_LZ: bool = if cfg!(target_feature = "thumb-mode") {
   |  ______________________^
69 | |     // ARM thumb targets have CLZ instructions if the instruction set of ARMv6T2 is supported. This
70 | |     // is needed to successfully differentiate between targets like `thumbv8.base` and
71 | |     // `thumbv8.main`.
...  |
77 | |     cfg!(target_feature = "v5te")
78 | | };
   | |_^
   |
   = note: see issue #49146 <https://github.com/rust-lang/rust/issues/49146> for more information
   = help: add `#![feature(const_if_match)]` to the crate attributes to enable

error[E0658]: `if` is not allowed in a `const`
  --> /home/ganshun/.cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.36/src/int/specialized_div_rem/mod.rs:68:22
   |
68 |   const USE_LZ: bool = if cfg!(target_feature = "thumb-mode") {
   |  ______________________^
69 | |     // ARM thumb targets have CLZ instructions if the instruction set of ARMv6T2 is supported. This
70 | |     // is needed to successfully differentiate between targets like `thumbv8.base` and
71 | |     // `thumbv8.main`.
...  |
77 | |     cfg!(target_feature = "v5te")
78 | | };
   | |_^
   |
   = note: see issue #49146 <https://github.com/rust-lang/rust/issues/49146> for more information
   = help: add `#![feature(const_if_match)]` to the crate attributes to enable

fotonick, 2020-09-18:
I can reproduce. I can also confirm that i'm using the versions that we locked in `make firsttime`:

    cargo --version  = cargo 1.44.0-nightly (8751eb301 2020-04-21)
    rustc --version  = rustc 1.44.0-nightly (b2e36e6c2 2020-04-22)
    rustup --version = rustup 1.22.1 (b01adbbc3 2020-07-08)

It looks to me like we haven't locked everything because during the `cargo build`, I see:

      Downloaded compiler_builtins v0.1.36

and this is where the compile error is. I don't understand why it's allowed to update compiler builtins when we've locked the compiler version—that seems dirty. Still, our easiest mitigation might be to update our nightly compiler version.

fotonick, 2020-09-18:
Extra weird. The latest release tag of `compiler_builtins` appears to be v0.1.35. How are we getting the new one?

Ignoring that for a moment, the easiest thing to try is to update our toolchain to the latest. Doing that, however, there's a massive binary size regression. `make mainboards` fails on the first board (`qemu-armv7`) with:

    Running `/usr/local/google/home/foton/src/oreboot/tools/layoutflash/target/x86_64-unknown-linux-gnu/debug/layoutflash target/arm-none-eabihf/release/fixed-dtfs.dtb target/arm-none-eabihf/release/image.bin`
    failed: File target/arm-none-eabihf/release/bootblob.bin is too big to fit into the flash area, file size: 6234767, area size: 524288

0.5MB to 6MB!? That's a heck of a regression. I'm going to stop my investigation here for now.

fotonick, 2020-09-18:
Ah. It's not necessary to push tags to github in order to release on crates.io. https://crates.io/crates/compiler_builtins shows 0.1.36. That's one mystery cleared up.

doriath, 2020-09-19:
In #278 I am adding both broken mainboards (emulation/qemu-armv7 and nuvoton/npcm7xx) to the "broken list" so that `make mainboards` works again. I am also adding `make mainboards` to CI, to ensure we catch new breakages before they are submitted.

Separately we should start debugging why those mainboards are broken and try to fix them,
* nuvoton/npcm7xx - I was also thinking about updating the toolchain to the latest to fix that one
* emulation/qemu-armv7 - this fails even on the old toolchain. I believe it was broken by #269 which actually added validation that the size of the binary fits into the flash area. The easiest fix would be just to increase the size of the area so that the binary fits, but as you said 0.5M to 6M is a big increase and I am not sure if that is the right fix.

fotonick, 2020-12-29:
Closing, as `make mainboards` builds without error now.

closed 2020-12-29 by fotonick