oreboot/oreboot #321: Fix write! and writeln! issues

doriath, 2020-10-27:
In this project we use `write!("...\r\n")` to print a line and the `\r` part is necessary (see https://github.com/oreboot/oreboot/pull/319/files#r512345413).

The above code causes clippy to issue errors and suggests using `writeln!("...\r")`, but it is pretty weird to see just `\r` at the end of the string.

I am planning to temporarily disable this check in #319 (inside Makefile) to avoid the errors and allow us to find better solution (which is the goal of this Issue).

Ideas:

* Talk to clippy to check if they have a solution if someone needs to use `\r\n` (is the expectation to have a string that ends with `\r`?). Maybe the check should no trigger if the string ends with `\r\n`.
* Create a new macro similar to `writeln` which will write a new lines as `\r\n` instead of just `\n`
* Go over the whole codebase and find uses of `write!` and `writeln!` and see if we can use the new macro instead
* (optional): It would be great to create some lint / check to ensure that we do not use `writeln!` or `write!` with just `\n` (also I am not sure how many exceptions we would have to such strong rule).

rjoleary, 2020-12-31:
I feel a good solution would be to have a `Console` Driver which re-writes the text passed through it. This driver would wrap the existing serial driver. Whenever a '\n' character is written to it, it'll write '\r' and '\n' to the underlying driver.

gogo2464, 2022-02-15:
For the check we could use a custom semgrep rule. If there is a more rustecean way I will do this.

Assign me to this issue.

orangecms, 2023-02-11:
We have done this in the new log library, inspired by https://github,com/dancrossnyc/r9. See also: #633 

Closing thusly.

closed 2023-02-11 by orangecms