oreboot/oreboot #371: OpenOCD config file for Hifive board link no longer good

MattWeatherford, 2021-02-14:
This link:

HiFive Unleashed config from: https://github.com/sifive/freedom-u-sdk/blob/master/bsp/env/freedom-u500-unleashed/openocd.cfg

no longer works

rjoleary, 2021-06-26:
Hi @MattWeatherford do you still find Oreboot support for the Hifive unleashed board important? We're thinking of dropping support in favor of the Unmatched. The support for the unleashed board was never working 100% properly.

MattWeatherford, 2021-06-26:
thank you for asking - the Unmatched board is the only one I am currently interested in - so yes, please prioritize that :) 
I would like to test whatever you come up with on my Unmatched hardware

rjoleary, 2021-07-01:
Here's the issue for adding the Unmatched support: https://github.com/oreboot/oreboot/issues/441