oreboot/oreboot #436: Import a crate to serialize Rust structs to C-style aligned byte slice

rjoleary, 2021-05-21:
There are a few unsafe blocks we can remove if we find a good create for this.

Two candidates are:

* https://github.com/xoac/endian_codec
* https://crates.io/crates/packed_struct

rjoleary, 2021-06-26:
I saw a blog post recently about this: https://adventures.michaelfbryan.com/posts/deserializing-binary-data-files/

daym, 2021-06-26:
ssmarshal is surprisingly good--but technically one is abusing the crate. 

I'm using zerocopy quite a lot otherwise--zerocopy is the right way to do it; but boy is it annoying.

rjoleary, 2021-06-30:
With the `core` library alone, is there a non-`unsafe` way to convert a `&[u8]` to a `repr(C)` struct?

fotonick, 2021-07-01:
The safe way I know how to do it is parse field-by-field like `nom` (supports `no_std`). You can also structure it to parse lazily with accessors (a la `capnproto`) if you want to avoid having original and deserialized versions in memory. These are not nearly as easy as transmutation.

TIL that there's a [safe transmute working group](https://github.com/rust-lang/project-safe-transmute).