oreboot/oreboot #441: Add support for Hifive Unmatched

rjoleary, 2021-06-26:
The is a feature request to add support for the Hifive Unmatched including:

* DDR initialization
* PCI initialization
* Booting Linux payload

We can probably drop support for the Hifive Unleased at the same time. The Unleashed is slower, more expensive and you cannot buy it any more.

rjoleary, 2021-07-01:
When support for Hifive Unleashed is dropped, also update all the documentation.

MattWeatherford, 2022-01-06:
Is anyone aware of any open source Hifive Unmatched code out there to do the DDR/PCI initialization that can be recycled in to oreboot?

Rwantare, 2022-01-06:
DDR initialisation is done [here](https://github.com/u-boot/u-boot/blob/master/drivers/ram/sifive/sifive_ddr.c)
and I don't think PCI is open source.  

I had a go at writing just the DDR bit for Oreboot, [here](https://github.com/Rwantare/oreboot/blob/unmatched/src/soc/sifive/fu740/src/ddr.rs) but for this to be tested I need to set up clocks and serial first, for which following the official docs wasn't sufficient.  

I haven't had time to look through how u-boot sets up the clocks to compare. Of course it could always be a bug in my implementation.

bwerthmann, 2022-03-03:
I have a Hifive Unmatched. Ping me if you need testing.

EmanuelLoos, 2025-01-26:
Hello @rjoleary and @githubisnonfree 

I have a HiFive Unmatched Board as well and could also do some testeng. How far is oreboot for the HiFive Unmatched?

orangecms, 2026-02-15:
for ref: https://review.coreboot.org/c/coreboot/+/76689

so you could also look at coreboot