oreboot/oreboot #443: Re-enable RISC-V QEMU Tests in CI

rjoleary, 2021-06-26:
See the `exit 0` skip here: https://github.com/oreboot/oreboot/blob/main/azure-pipelines.yml#L133

These CI checks were disabled because they re-build QEMU from source each time which added an unnecessary 8 minutes to the tests. Instead, we should be using a pre-built QEMU or caching the build.

derpsteb, 2021-12-22:
Could use [this](https://github.com/actions/cache) caching action. 
Would requiring adding another step to the job. Then the other jobs could check if there is a cache hit before running.
Initial ideas for a cache-key could be: The built binary or the Makefile. Both only exist if at least one build has run. The Makefile potentially wouldn't catch version changes.

orangecms, 2022-02-07:
Note: We have switched to GitHub Actions; now see
https://github.com/oreboot/oreboot/blob/main/.github/workflows/fsp-qemu.yml#L72

closed 2022-02-08 by orangecms