oreboot/oreboot #445: Poll: Is it "oreboot", "Oreboot" or "OreBoot"?

rjoleary, 2021-06-26:
Add an emoji with your vote:

* :rocket: for "oreboot"
* :heart: for "Oreboot"
* :tada: for "OreBoot"

rminnich, 2021-07-05:
when does the poll close

ArthurHeymans, 2021-07-05:
With coreboot, it was decided that it should always be full lowercase for trademark reasons. IANAL, so I comment on the validity of that decision. However if you start a phrase with a lowercase letter it looks like you made an typo. If that trademark reason is a thing and if you have to pick, I'd suggest at least going uppercase for the first letter as that is less confusing.

rjoleary, 2021-07-08:
> when does the poll close

Let's close it in about a week.

rjoleary, 2021-07-08:
Could this be another option: lowercase except at the beginning of a sentence?

rjoleary, 2021-07-22:
The poll is closed. The final results are:

* 11 for "oreboot"
* 5 for "Oreboot"

closed 2021-07-22 by rjoleary