oreboot/oreboot #446: Discussion: Should oreboot support embedded systems?

rjoleary, 2021-06-26:
The two options are:

1. Oreboot supports just PC-like systems.
   * Example: laptops, desktops, servers, smart phones, rpi, anything that can boot Linux, ...
2. Oreboot supports all of the above *and* embedded system.
   * Example: low-powered microcontrollers, EC, systems with < 1MiB memory, OpenTitan, ...

One downside of embedded system is they bring additional challenges and often need once-off drivers or application logic.

fotonick, 2021-06-28:
What are the architectural or governance considerations? For the embedded use case, anyone could add their own mainboards directory with whatever one-off drivers they wanted, right? Is there any common component of oreboot that would feel the squeeze if we suddenly had strict memory requirements?

alistair23, 2021-06-29:
I think a good question is what embedded system do you foresee that wants/uses a boot loader?

Usually embedded systems don't have any resources to spare on a bootloader. As well as that they generally don't need something like Oreboot as they just have a custom boot rom to support the boot process. If there is a use case for Oreboot then I think it makes sense to support.

jhand2, 2021-07-02:
As someone who is fairly new to oreboot, I think the best way to answer this question is to answer "what does oreboot that is not board specific?". And then "Can an embedded platform use these things unmodified?"

OpenTitan is an interesting example because for a security chip there are likely a lot of board specific things make it a more needy platform (initializing memory protections, setting up key hierarchies, special secure-boot flows, etc). Still it would be possible to stash this stuff in board-specific code as long as the common oreboot code provides enough value (over writing something custom).

rminnich, 2021-07-02:
it should support any system where it is useful. If that includes embedded, then I'm for it. I'm working on beaglev now.