oreboot/oreboot #447: Measure unit test coverage in CI

rjoleary, 2021-06-26:
There are some instructions on setting this up for Azure Pipelines here: https://docs.microsoft.com/en-us/azure/devops/pipelines/tasks/test/publish-code-coverage-results

Documentation from the Rust compiler: https://doc.rust-lang.org/beta/unstable-book/compiler-flags/instrument-coverage.html

gogo2464, 2021-12-04:
I want to be assigned for this issue.

I have more epxerience with travis-ci than azure devops (tough I learn it at school). Can I use it instead?

gogo2464, 2021-12-28:
I saw an azure code. Then I requested a Request Free Parallelism to azure devops. I will use azure devops.

gogo2464, 2022-01-15:
@rjoleary , @rminnich  ready for a review. Check  https://github.com/oreboot/oreboot/pull/537

gogo2464, 2022-02-05:
As told in this PR https://github.com/oreboot/oreboot/pull/527#pullrequestreview-847016588 we migrated to travis ci/cd.

Merged! https://github.com/oreboot/oreboot/pull/537 Fell free to close this issue.

closed 2022-02-05 by orangecms