oreboot/oreboot #456: Avoid duplicating assembly for different boards

jhand2, 2021-07-08:
We should make an effort to deduplicate assembly code that is needed for different use cases but is largely the same. One such example is code in `src/arch/x86/x86_64/src/bootblock.S` and `src/arch/x86/x86_64/src/bootblock_nomem.S`. They contain identical code for entering long mode, ISA constants, and some other things. Maintaining assembly is a pain, it's best to reduce the burden as much as possible.

My recommendation would be to just run assembly files through the C preprocessor to enable the use of `#include` and `#define`. I'm happy to work on this, just want to post it here to make sure folks agree with the approach.

rjoleary, 2021-07-08:
This would be a good improvement! It could bite us if we change one file but forget to change the other.

Just to cover the bases, are there any solutions besides the C preprocessor?

jhand2, 2021-07-09:
I considered something with string replace, along the lines of

sample.asm
```
function:
  {asm_to_include}
  ...
  ret
```

extras.inc
```
movq ..., ...
```

main.rs
```
global_asm!(include_str!("sample.asm"),
            asm_to_include = include_str!("extras.inc"));
```

In theory this should work but:
1. I couldn't actually get it to work. For some reason the string replacement doesn't quite work right.
2. You have to so some extra work every time you include some assembly. This is maybe not so bad if you just wrap arch-specific code in a rust library so you don't include the same asm in multiple places.

jhand2, 2021-07-09:
Oh also this is a decent option so we have something in rust rather than calling the C preprocessor as a subprocess: https://docs.rs/gpp/0.6.0/gpp/

jhand2, 2021-07-13:
I wrote a minimal preprocessor from scratch here: https://github.com/oreboot/oreboot/pull/458

It's all rust and can easily be called from build scripts without any external dependencies.