oreboot/oreboot #457: we need an automated fix for clippy warnings

rminnich, 2021-07-08:
Right now, if clippy finds issues like excessive parens, it will complain. 

But no invocation of clippy fix will fix it. 

We need a way to fix ALL warnings that clippy generates.

rjoleary, 2021-07-08:
https://stackoverflow.com/questions/53670348/can-rusts-clippy-do-autocorrection-autofix shows multiple possibly ways to auto fix clippy issues.

Perhaps:

```
First you need to install clippy nightly (rustup component add clippy --toolchain nightly-x86_64-unknown-linux-gnu), and after that run it with: cargo +nightly clippy --fix -Z unstable-options
```

Or perhaps try this if it's now stable:

```
cargo clippy --fix
```

rjoleary, 2021-07-08:
This seemed to work for me:

```
diff --git a/Makefile b/Makefile
index 88c99cda0c..40a269d3e9 100644
--- a/Makefile
+++ b/Makefile
@@ -116,7 +116,7 @@ BROKEN_CRATES_TO_CLIPPY := \
 # TODO: Remove write_with_newline
 CRATES_TO_CLIPPY := $(patsubst %/Cargo.toml,%/Cargo.toml.clippy,$(filter-out $(BROKEN_CRATES_TO_CLIPPY),$(CRATES)))
 $(CRATES_TO_CLIPPY):
-       cd $(dir $@) && cargo clippy -- -D warnings -A clippy::write_with_newline
+       cd $(dir $@) && cargo clippy --allow-dirty --fix -Z unstable-options -- -D warnings -A clippy::write_with_newline
 .PHONY: clippy $(CRATES_TO_CLIPPY)
 clippy: $(CRATES_TO_CLIPPY)
```

* `--fix`: Automatically fixes clippy issues
* `-Z unstable-options`: Because of the error "Usage of `--fix` requires `-Z unstable-options"
* `--allow-dirty`: Because the error "the working directory of this package has uncommitted changes, and `cargo fix` can potentially perform destructive changes; if you'd like to suppress this error pass `--allow-dirty`, `--allow-staged`, or commit the changes to these files:"

orangecms, 2025-09-25:
Tooling has matured a lot.

Closing now.

closed 2025-09-25 by orangecms