oreboot/oreboot #497: Assembly preprocessor (rpp) improvements

jhand2, 2021-10-29:
rpp does not currently support multiline macros such as
```
#define MACRO some \
    stuff
```

This should be an easy thing to add

pdogr, 2022-01-25:
Hi!
I've tried adding this feature in #538. I was not sure if we should preserve the exact structure (escaped newlines) in macro definition, though this would really depend on whether the assembly has a free-form syntax?

jhand2, 2022-01-26:
Thanks! It's worth noting that since https://github.com/oreboot/oreboot/issues/504 was resolved, oreboot assembly no longer uses rpp to insert macros. Apologies for not updating this issue.

That says, rpp is still fun endeavor and I'll take a look at the PR.

pdogr, 2022-01-27:
Sorry! I should've asked before picking this up.

jhand2, 2022-01-27:
No problem :smile: I might move rpp into it's own repo/crate at some point.