oreboot/oreboot #500: qemu-q35 debug port print breaks boot

derpsteb, 2021-11-01:
When running `make run` for qemu-q35 I can not get past the first print onto the debug port [here](https://github.com/oreboot/oreboot/blob/main/src/mainboard/emulation/qemu-q35/src/main.rs#L30-L32). 

When this line is commented out I can at least get up to loading the payload (L47), at which a panic will occur and the following message will be printed forever:
```
Welcome to oreboot                                                                                       
boot flag is ffff, not aa55, at ffff,                                                                                                                                                                              
                                                                                                                                                                                                                   
PANIC: Welcome to oreboot                                                                                                                                                                                          
boot flag is ffff, not aa55, at ffff,                                                                                                                                                                              
                                                                                                         
PANIC: Welcome to oreboot                                                                                                                                                                                          
boot flag is ffff, not aa55, at ffff,
```

Hopefully I will be able to resolve it myself, but maybe someone knows whats wrong here :)

derpsteb, 2021-11-01:
This is the content of SetupHeader, before it throws the error:
```
SetupHeader: SetupHeader {                                                                               
    setup_sects: 0xff,
    root_flags: 0xffff,                                                                                  
    syssize: 0xffffffff,
    ram_size: 0xffff,
    vid_mode: 0xffff,
    root_dev: 0xffff,
    boot_flag: 0xffff,
    jump: 0xffff,
    header: 0xffffffff,
    version: 0xffff,
    realmode_swtch: 0xffffffff,
    start_sys: 0xffff,
    kernel_version: 0xffff,
    type_of_loader: 0xff,
    loadflags: 0xff,
    setup_move_size: 0xffff,
    code32_start: 0xffffffff,
    ramdisk_image: 0xffffffff,
    ramdisk_size: 0xffffffff,
    bootsect_kludge: 0xffffffff,
    heap_end_ptr: 0xffff,
    ext_loader_ver: 0xff,
    ext_loader_type: 0xff,
    cmd_line_ptr: 0xffffffff,
    initrd_addr_max: 0xffffffff,
    kernel_alignment: 0xffffffff,
    relocatable_kernel: 0xff,
    min_alignment: 0xff,
    xloadflags: 0xffff,
    cmdline_size: 0xffffffff,
    hardware_subarch: 0xffffffff,
    hardware_subarch_data: 0xffffffffffffffff,
    payload_offset: 0xffffffff,
    payload_length: 0xffffffff,
    setup_data: 0xffffffffffffffff,
    pref_address: 0xffffffffffffffff,
    init_size: 0xffffffff,
    handover_offset: 0xffffffff,
    kernel_info_offset: 0xffffffff,
}
```

This seems wrong to me Oo

orangecms, 2021-11-01:
Commenting out the entire declaration of debug, its init and the pwrite, I do get that payload to run, with the Linux kernel eventually panicking.

With a kernel from Arch, I get an endless loop of

```
Welcome to oreboot
Kernel offset is 3e00
Copy ff003e00 to 1000000 for 300000
Running payload
on to 0x1000200 Welcome to oreboot
```

derpsteb, 2021-11-14:
That looks very similar to what I am seeing.

derpsteb, 2022-01-08:
Looked at it again today with @jhand2 and after commenting out [this](https://github.com/oreboot/oreboot/blob/main/src/mainboard/emulation/qemu-q35/src/main.rs#L41-L43) line a kernel boots.

derpsteb, 2022-01-12:
The problem seems to be related with the timer code [here]( https://github.com/oreboot/oreboot/blob/main/src/drivers/uart/src/debug_port.rs#L38-L42). Removing it makes everything work fine.

orangecms, 2022-01-12:
Interesting! That could also be related to the issue we have with the A300 port.

PR #535 (2022-01-15): Initialize HPET driver in DebugPort
Add a function `enable()` to the HPET driver. This allows the calling driver, `DebugPort` in this case, enable the timer if it wants to use it. This is necessary for the qemu-q35 board as it does not init the timer, like the romecrb board for example.
This fixes #500.

I was contemplating whether `hpet0()` should do this automatically, but as the code is already used in other boards and there is a comment saying the caller should ensure enablement I felt like this is sensible solution. Happy to hear other opinions.
closed 2022-01-15 by derpsteb