oreboot/oreboot #504: Replacing Assembly Preprocessor with Rust Built-in Features

retrage, 2021-11-07:
I'm new to this project, so correct me if I missed something.
As far as I know, PR #458 introduced rpp for preprocessing assembly files. The primary motivation is to improve assembly code reusability because Rust does not provide its macro system to assembly.

However, I cannot find any reason to employ the C preprocessor-like syntax for the project written in Rust. I think it is better to use rich Rust's compile-time features for preprocessing. The current primary usage of `preprocess_asm!()` is to define constant value macros and include other files. It can be replaced by nightly Rust's built-in feature because `global_asm!()` accepts named parameters. Here is my PoC:
https://github.com/retrage/oreboot/tree/asm-const-poc

The macro-defined values are defined in [src/consts/asm.rs](https://github.com/retrage/oreboot/blob/asm-const-poc/src/arch/x86/x86_64/src/consts/asm.rs). The macro references are replaced with Rust named parameters (e.g. `Pse` -> `{pse}`).
The corresponding const values are given as arguments of `global_asm!()`.
```rust
global_asm!(
    include_str!("../../../../arch/x86/x86_64/src/bootblock_nomem.S"),
    pse = const asm::PSE,
    // snip
    options(att_syntax)
);
```
This design also allows us to use other Rust const features like `const fn` without additional efforts.

The main drawback of this idea is that the number of `global_asm!()` arguments may be large, and the code looks redundant. We can hide the arguments by defining a macro that uses default values.

I can provide PR with more improvements if anyone is interested in my idea.

orangecms, 2021-11-07:
Oh wow, that's great, thank you a lot! It makes maintenance a lot easier with otherwise duplicated definitions. We currently have a bug with kernel execution in the q35 QEMU test port, which I suspect to be related to some divergence. With your ideas here, that would be totally mitigated. :)

rminnich, 2021-11-07:
Nice to have you here. Please submit your PR. We are still learning and welcome PRs like this!

jhand2, 2021-11-07:
I'm not overly attached to rpp, so ya if there's a more idiomatic rust way to do this I'm all for it 😊

jhand2, 2021-11-07:
Caveat that I'm a little weary of using nightly rust features because it would be nice to move to a stable tool chain. I don't know enough about this feature to know if/when it's headed to stable.

retrage, 2021-11-08:
Thanks, so I'll submit a PR.

retrage, 2021-12-03:
Close this issue because my PR #513 is merged.

closed 2021-12-03 by retrage