oreboot/oreboot #533: start development for Rockchip

orangecms, 2022-01-09:
With sunxi (Allwinner), we have introduced support for SoCs / boards from well-known vendors who's chips are used in many devices, such as Orange Pi, PinePhone, TV boxes, etc.. The kickstart happened via the `xfel` tool from the xboot project.

For Rockchip, we can start in a similar fashion, using [rkflashtool](https://github.com/linux-rockchip/rkflashtool) and/or [xrock](https://github.com/xboot/xrock).

hanetzer, 2022-01-19:
one annoyance: you need to process the bootblob in a
certain manner. coreboot has a python script that does this.
And last I looked u-root's kexec tool can't load dtb files?

orangecms, 2022-01-19:
Good to know, I'll have a look at that script and also check U-Boot.

Loading a dtb shouldn't be an issue, we could add that if necessary. :-)

hanetzer, 2022-01-22:
for the record: I own both a rk3288 and rk3399 based  chromebook, with the servo debug headers installed,
along with a servo debug board, so I have *a lot* of debug capabilities in this area.

orangecms, 2022-01-27:
Excellent! I have a ROCK64 that I wanted to use for initial development.
However, I need to figure out how the USB modes work, can you help with that @hanetzer?

`rkflashtool` is said to not support mask ROM mode fully, only bootloader mode: https://wiki.radxa.com/Rock/flash_the_image#Linux
However, as per this help text, it can load DDR init and USB loader from mask ROM mode: http://rockchip.wikidot.com/rkflashtool

Do you have experience with that?

jwinarske, 2022-08-01:
This is a good working reference for RK3399:
https://gitlab.com/DeltaGem/levinboot

I'm interested in getting Oreboot running on Pine phone Pro - RK3399, as well as other targets.  I've been studying levinboot.

felixsinger, 2022-12-29:
https://gitlab.com/pine64-org/quartz-bsp/rkdeveloptool

orangecms, 2025-09-25:
Noting down here that we have initial support for the RK3566 and I started reversing the DRAM init, being at ~80%.

Code can be loaded over USB for development, requiring no extra header. See the RK3566 README and Makefile. :)