oreboot/oreboot #568: Is qemu-riscv available now?

YdrMaster, 2022-04-21:
I have tried to boot rustsbi-qemu with oreboot but it didn't work. Is there any documentation that can guide me, or is it not available yet?

orangecms, 2022-04-22:
Hi, thanks for the inquiry!

QEMU with RISC-V support was just released a few days ago:
https://www.qemu.org/2022/04/19/qemu-7-0-0/

We will see that we set up a board in `src/mainboard/emulation/qemu-riscv` with SBI. Would you like to contribute to that?

Note:
Currently, we only have a bit of SBI code so that it just works for the Nezha (`src/mainboard/sunxi/nezha`). That code use some refactoring. :-)

YdrMaster, 2022-04-26:
Thanks for your answer!

orangecms, 2022-07-20:
I realized the QEMU 6 also had RISC-V support already. Please try this here out on your machine @YdrMaster:
https://github.com/u-root/cpubinaries/pull/7

bwerthmann, 2022-11-07:
@YdrMaster please see #622.

closed 2022-11-07 by bwerthmann