oreboot/oreboot #598: DTB (FDT) parsing & patching

orangecms, 2022-08-08:
For dtb based platforms, we need the ability to pass memory configuration (DRAM size, notably), which is determined at runtime.

We do have a small library around device trees already for the DTFS. More:
- https://github.com/repnop/fdt
- https://github.com/rustsbi/serde-device-tree

orangecms, 2022-08-16:
related: #498

rmsyn, 2023-01-07:
FWIW my vote is for https://github.com/repnop/fdt, no dependencies, does the job.

orangecms, 2023-01-21:
https://github.com/rivosinc/salus/tree/main/device-tree/src

also for reference

orangecms, 2025-04-24:
another implementation in R9: https://github.com/r9os/r9/blob/main/port/src/fdt.rs

the current https://github.com/oreboot/oreboot/blob/f6a3e09d8f3f2f525c8961907f01797d00e435d5/src/lib/layoutflash/Cargo.toml uses repno/fdt

orangecms, 2025-04-24:
for [dtvis](https://github.com/platform-system-interface/dtvis), we are currently using https://github.com/platform-system-interface/device_tree-rs, forked from another implementation again