oreboot/oreboot #626: Address deprecation of Node.js 12 in CI tasks

bwerthmann, 2022-11-02:
```
Node.js 12 actions are deprecated. For more information see: https://github.blog/changelog/2022-09-22-github-actions-all-actions-will-begin-running-on-node16-instead-of-node12/. Please update the following actions to use Node.js 16: actions/checkout, actions/checkout
```

https://github.com/oreboot/oreboot/actions/runs/3371817811

---
Initial Thinking:

* Update CI Actions to use `actions/checkout@3`

bwerthmann, 2022-11-07:
Looks like I missed one:

> Node.js 12 actions are deprecated. For more information see: https://github.blog/changelog/2022-09-22-github-actions-all-actions-will-begin-running-on-node16-instead-of-node12/. Please update the following actions to use Node.js 16: codecov/codecov-action

orangecms, 2022-11-07:
Whoopsies! 80/20? :D

closed 2022-11-07 by bwerthmann