oreboot/oreboot #639: Chromebook support

rmsyn, 2022-11-14:
Hi, I recently started looking at supporting Chromebook builds in oreboot.

Maybe following a similar pattern to #533, except with the need to include an Intel binary blob (looking at an x86_64 mainboard).

I have a Chromebook CB311-9H (octopus-bobba) piece of hardware to test on, and have started porting the C code from `coreboot` in my [cb311](https://github.com/rmsyn/oreboot#cb311) branch.

So far, just have the EC ACPI ASL files copied over, and some of the EC code ported. Will need to add some parser code in a `build.rs` using the [acpi](https://github.com/rust-osdev/acpi) crate.

Before I go too much further, is this something that `oreboot` wants to pursue? I'm happy to take this on as a project, and shouldn't need too much oversight.

Mainly just don't know how more experienced project members feel about the binary blob...

rminnich, 2022-11-15:
wow. Cool. But a question. Have you considered an ARM chromebook, which would be much more open source? but if you still want to do this, you are most welcome here!

rmsyn, 2022-11-15:
> Have you considered an ARM chromebook

Definitely, I had one at one point, and wouldn't mind getting one again for this project. The stuff I'm working on now is for the EC (common to all Chromebooks, I think).

I'll probably be doing this in stages (EC support, mainboard, bobba variant, etc). When I get closer to having something ready for review, I'll ask about how you want me to split it up. Who knows, I may have an ARM Chromebook by then :)

ElonGaties, 2024-07-08:
hallo, just wanted to check in on the progress or possible completion of getting oreboot working on your bobba chromebook, I was personally considering trying to get oreboot working on my phaser chromebook (also in the octopus line of chromebooks)

rmsyn, 2024-07-09:
I have basically abandoned efforts on Chromebook support.

If you are working on porting oreboot to an `x86_64` Chromebook, you'll need to resurrect the EDK support for initial board bring-up.

You may also need to do some modernizing on the wrappers, since the EDK support was dropped a while ago.

Best of luck on your journeys :)