oreboot/oreboot #641: aarch64 startover

orangecms, 2022-12-06:
see https://github.com/rust-embedded/aarch64-cpu

Cortex-A has good support in Rust Embedded already. We can build on top of that and port something like a Rockchip or Allwinner ARM SoC. See also #533.

orangecms, 2023-02-20:
see #684 for Allwinner H616

felipebalbi, 2024-03-01:
Hi,

I would like to help, where possible, with reworking the AArch64 port. It appears that fixing what currently sits in `src.broken` is not the way to go. Could we get some better description of what the community is wanting to do for the AArch64 port?

Thanks

orangecms, 2024-03-03:
Hi and welcome to the project!

There is not much of an architecture specific goal for oreboot, so it's your chance to contribute and discuss it, if you feel like there is a need. :)

The general idea is to keep things simple and only do as much as necessary to boot up a kernel without any sort of runtime services a fanciness. We leave it up to the OS or bootloader to do more. See also [the notes on the boot flow](https://github.com/oreboot/oreboot/blob/main/Documentation/boot-flow.md). That usually boils down to some GPIOs, a UART for debugging, DRAM init, loading a kernel to DRAM and jumping to it. Depending on the platform, that may mean some extra code in order to load the kernel from some storage in the first place, e.g., SPI flash, eMMC or SD card.