oreboot/oreboot #681: why is some code missing？

Mammon-github, 2023-02-10:
Doucument say there’s a folder ”src/sifive/hifive“ ，why is it missing？it also happends on other device in doc.
And i also noticed that in nezha，if use xfel to init the DRAM，why the Bt0 still there？that make me very confused.
In "src/soc/sifive/fu540", the bootblock.S called a function "_start_boot_hart",  i can't find it in the project ,maybe it's provided by some crates , and it's very difficult to me to find them.
Sorry about ask so many questions, thank you for answering my confusion.

Mammon-github, 2023-02-10:
Another question，when did the xtask be called？ In bt0 i knew it fills the main_stage_head，bt0 use those information to copy the main_stage to ddr from flash, but how xtask did this? it seems like xtask worked before bt0，And i don‘t think i see a xtask function called before copy the main_stage in bt0.
ps: i am a Chinese guy, so my Engilsh may have some mistakes, please forgive me.

orangecms, 2023-02-10:
> Doucument say there’s a folder ”src/sifive/hifive“ ，why is it missing？it also happended on other device mentioned in doc. And i also noticed that in nezha，if use xfel to init the DRAM，why the Bt0 still there？that make me very confused. In "src/soc/sifive/fu540", the bootblock.S called a function "_start_boot_hart", i can't find it in the project ,maybe it's provided by some crates , and it's very difficult to me to find them. Sorry about ask so many questions, thank you for answering my confusion.

Hi, welcome to the oreboot project :)

Yes, we have some things in the docs that correspond to code that has been removed to get forward with the development. That can still be found in the history and is described here:
https://github.com/oreboot/oreboot/blob/main/graveyard.md

Overhauling the README and docs is an open task to do. Sorry for the confusion!

orangecms, 2023-02-10:
> Another question，when did the xtask be called？ In bt0 i knew it fills the main_stage_head，bt0 use those information to copy the main_stage to ddr from flash, but how xtask did this? it seems like [...]

When you run `make flash` in the Nezha board directory, it will run xtask under the hood, and
- build what is currently called `bt0`
- add the header to it (`eGON` etc)
- build what is currently called `main`
- compose an image with the above
- optionally add your device tree and payload
Does that answer your question?

closed 2023-02-13 by Mammon-github