oreboot/oreboot #700: Could you consider commiting the lockfile to the repo ?

happysalada, 2023-07-09:
That would make it much easier to package this for linux distributions.
Thank you for all the work that went into this!

orangecms, 2023-07-09:
Oh, I thought we had it.... dang, thank you for noticing! It explains a few things. :sweat_smile:

nyabinary, 2024-10-20:
> Oh, I thought we had it.... dang, thank you for noticing! It explains a few things. 😅

Any update? :P

orangecms, 2024-10-21:
Hey sorry we haven't yet done this. I'll see with another developer soonish - up until recently, I had been the only one, so the project got stalled somewhat.

To be honest, I'm actually surprised anyone is using oreboot at this point - could you help me understand a bit how? We haven't much discussed distribution matters in the project yet, and I am a bit split between going for a rather opinionated approach vs offering flexibility, mostly because the latter comes with complexity. I hope we can offer what you need.

closed 2024-10-24 by orangecms

happysalada, 2024-10-27:
Hey thank you so much for adding the lockfile to the repo ?
Will the project work with releases ? 
Would it be possible to have a github release ?
It's of course no problem if it's not right now.
But when you feel the timing is right, having a github release would make it much easier for downstream package maintainers to package oreboot.
Thank you!

happysalada, 2024-10-27:
Right, the binary will likely need to be compiled and distributed right ?
I am perhaps a little uninformed on this, but i thought there would be potential flags for customisation in the binary. Then from another computer someone could just build the package, put it on a usb and use it to boot a new machine. So the first computer would just have a distro that enables you to build the binary.
When you mention an image, im getting the sense that it wont be just a simple cargo build with features but perhaps a more complicated process.
Nonetheless, the resulting images could be available in various distros and then people would move them to a usb storage or other to boot another machine.
I havent thought about any of this though, i was just thinking of trying to package this to learn more about the initiative since it looks interesting

orangecms, 2024-10-28:
Mind that there is no single binary in any case.
The issue with distributing images is that modifying a given images requires stable protocols and extra tools. It is far more trivial to build everything from source. Otherwise, it is a lot of effort, and we'd rather put that into moving the project forward in other aspects. I will gladly assist if you want to work on such things, though.

happysalada, 2024-10-28:
Thank you so much for your kind answer. Let me level up a bit and come back to you after that.

PR #758 (2024-10-24): Rust upgrade, commit Cargo.lock
resolves #700