oreboot/oreboot #707: Community Chat

m0veax, 2023-08-19:
Hi all,

not really an issue, but I found myself diving deeper into oreboot and wondered if there is a matrix channel or something like that to drop a quick question and help each other out.

I made the experience, that I learn a lot by just lurking around and read questions and answers from other people too. If you agree, I would even do the work and create an unencrypted matrix channel for us. We could add it to the README if it is some kind of populated :)

Best regards

Patrick

rminnich, 2025-02-28:
we've had the slack channel since 2019, with all discussion history there. Why not that?

orangecms, 2025-09-25:
We still need to add to the README that we're in the OSFW Slack.

There has been an issue with the invite page; it is currently beig worked on.

closed 2025-10-03 by kanakshilledar