oreboot/oreboot #741: Milk-V Duo (CVITek SoC) support

orangecms, 2024-04-19:
As of now, those SoCs are not fully documented and only binary blobs avilable.

For reference:
- https://github.com/milkv-duo/duo-files (datasheet and manual)
- https://github.com/milkv-duo/duo-buildroot-sdk/tree/develop/fsbl/plat/cv180x/bl2_objs/cv1800b_sophpi_duo_sd/bl2
- https://github.com/milkv-duo/duo-buildroot-sdk/tree/develop/middleware/v2/cv180x/ko

orangecms, 2024-04-19:
for the Duo S, the DDR controller registers are documented, just not how to use them

the SPI flash is memory mapped
EMMC DAT 0 and 3 determine the boot source
dat3 down, dat 0 up means boot from SPI NOR flash

cyrozap, 2024-08-31:
Looks like the FSBL has been open source (or at least "source available" if not explicitly licensed) since May: https://github.com/milkv-duo/duo-buildroot-sdk/commit/63fa9d586413b92bec81d95997695646719a8045

orangecms, 2024-09-01:
Ah yes I've actually translated that code, am waiting for a reply to my request for a license.

See my fork:
https://github.com/orangecms/oreboot/tree/all-the-things-wip/src/mainboard/milk-v/duo-s

cyrozap, 2024-09-02:
> ... am waiting for a reply to my request for a license.

Do you have a plan for if you don't get a response, or if you get a negative one? Maybe we could document what the FSBL does and use that documentation to re-implement it cleanly (both legally and in terms of code quality).

orangecms, 2024-09-14:
What the FSBL does is writing thousand of values to registers.
I have no plans and IANAL...

For reference: https://github.com/sophgo/fsbl/issues/1

cyrozap, 2024-09-15:
> What the FSBL does is writing thousand of values to registers.

Sorry, I should have been more specific. What I meant by "document what the FSBL does" is that we could read the FSBL code and then describe in human language (or a simple scripting language) the sequence of steps the FSBL takes to initialize the chip, including what values to write to which registers and the order in which to write them. [Andrew "bunnie" Huang](https://en.wikipedia.org/wiki/Andrew_Huang_(hacker)) describes something like this in his 2014 blog post, [From Gongkai to Open Source](https://www.bunniestudios.com/blog/2014/from-gongkai-to-open-source/) (the relevant section is titled "Booting an OS"). In that post, he explained how he extracted just the uncopyrightable facts from proprietary chip initialization source code, to ensure that the bootloader he was writing that was derived from those facts would be able to be freely-licensed. I think it should be possible to do something similar with the FSBL code, where the sequence of register writes could be rewritten in a stripped-down macro language (or described in a CSV or text file, or just some limited "write value X to register Y" Rust code) to ensure that only the facts are being used in the resulting code.

> I have no plans and IANAL...
>
> For reference: https://github.com/sophgo/fsbl/issues/1

Considering it's been about five months since you filed that issue and there's been no response from Sophgo, I think we should proceed under the assumption that the FSBL code will never be properly FOSS-licensed. In other words, either we document the facts of the initialization process and write our own FOSS code based on those facts, or we say "YOLO" and directly port Sophgo's code without regard for the license (or the lack of one).

orangecms, 2025-01-12:
> I think it should be possible to do something similar with the FSBL code, where the sequence of register writes could be rewritten in a stripped-down macro language (or described in a CSV or text [...]

I already simplified the code a lot and wrote it in a more concise manner. Could you please read it and say from your side whether it meets what you propose?

kanakshilledar, 2025-04-25:
Is it still a WIP port?

orangecms, 2025-04-25:
> Is it still a WIP port?

yes, see the referenced repo and what's (not) in oreboot upstream

pftbest, 2025-08-19:
Thanks!
As I see it, there were a lot of other small fixes since that patches, including update to the new nighly version, so I would rather stay on the latest branch. The memory seems to work, and if the `rx_dll_code dly_sel` errors are not critical, then it's fine by me.

pftbest, 2025-08-19:
I've run into some issue, the memory itself works ok, all 512MB seems to be fine, but while I was testing it I noticed that it runs quite slow. I made a tiny SBI application to benchmark it [rv_sbi.bin](https://github.com/pftbest/rv_sbi/blob/main/rv_sbi.bin)

I can run it both with oreboot over usb and I can make a `fip.bin` to run from the sd-card with old fsbl, but with the same rustsbi like this:

```bash
./fiptool \
    --fsbl data/fsbl/cv181x.bin \
    --ddr_param data/ddr_param.bin \
    --opensbi ../oreboot/target/riscv64imac-unknown-none-elf/release/milk_v-duo_s-main.bin \
    --uboot ../rv_sbi.bin --rtos ../rv_sbi.bin
```
For me it runs 33 times slower with oreboot compared to old fsbl. Is it possible some icache is not enabled or some PLL runs at different speed? I tried to check with an older commit as you suggested, but I get compile errors on most of them, and if I go back in time even more, it compiles and runs, but my application starts crashing on load.

orangecms, 2025-09-22:
I was just hinted on this:
https://github.com/ParkerTenBroeck/milkv-duos-rs

just leaving it here as a note