oreboot/oreboot #751: DTFS design, pluggability

orangecms, 2024-06-23:
I need help designing our pluggability model. The DTFS is a nice idea - but it only works for very narrow scenarios.
What I find myself doing is adjusting it all the time or hacking into the setup around it to cut down images to fit in SRAM etc..
The following scenarios and variants therein are common:
- image for SRAM only, either with a simple payload or a payload with a DTB, or loading a payload from storage
- image for flash, SD card, eMMC... they sometimes have different layout requirements, but those are vendor specific

Currently, the DTFS we have is only useful for the second case. And for the first one, I could now go ahead and add a handful of DTFS files to covert their scenarios.
BUT:
- SRAM is small and loaders are slow; it is desirable to keep the image as small as possible
- payload with and without DTFS already requires full duplication, with just one addition; nasty to keep in sync
- I have scenarios where I want to _build_ the payload; we have no suitable setup yet

I still feel the same way that I mentioned at some point before: I would prefer the DTFS to not stem from a separate file, but rather generate it as part of the build process, since only in the process can we really tell what the optimal layout would be.

In addition, there are scenarios where a single layout is not suitable.
Think of the modern AMP (asynchronous multi-processing) SoCs.
Some processors have cores that do not support the same ISA, have different modes, different cache sizes, etc.. They still boot as single systems though.
Example:
The StarFive JH7110 has 4 +1 + 1 cores. In the main U74MC, 4 U7 RV64GC harts are SMP-like, all the same, RISC-V with M, S and U modes. Then there is one more RV64IMAFDC hart (the MC in U74MC is apparently for "management core") which is an S7, offering only M and U modes. I..e, you could e.g. run a 4-hart Linux + a single hart Zephyr. Then there is an additional, RV32xxx hart somewhere else, which can be instrumented by an OS later, but _could_ also be provisioned by firmware already.
See also the [OpenAMP project's approach](https://www.openampproject.org/news/system-dt-lopper).

Idea?

ArthurHeymans, 2024-06-23:
So how about generating the DTFS FDT file based on some heuristics?

orangecms, 2024-06-23:
> So how about generating the DTFS FDT file based on some heuristics?

Yea that is essentially what I'm thinking.
And what works is vendor/SoC specific, unfortunately, so it's hard to bake that into a general framework. I'd instead provide helpers and add some guidance.

orangecms, 2025-10-03:
After a bunch of iterations and such, I have a clearer idea now. :)
To get there, I have a good bunch of PRs pending / coming up, starting with some cleanups/fixing:

- #805 
- #798 
- #799
- #800

and more to come!