oreboot/oreboot #761: maybe unsound in print_mem and print_strmem

lwz23, 2025-01-02:
Hello, thank you for your contribution in this project, I am scanning the unsoundness problem in rust project.
I notice the following code:
```
pub fn print_mem<T>(s: *const T) {
    let p = s as u32;
    let m = unsafe { core::ptr::read_volatile(p as *mut u32) };
    print_hex(m);
}

pub fn print_strmem(s: &str) {
    let p = s.as_ptr();
    let m = unsafe { core::ptr::read_volatile(p as *mut u32) };
    print_hex(m);
}
```
 ** Pointer alignment problem ** :
Pointers of type - '&str' may not meet the 4-byte alignment requirements of type u32.
- When you convert the '&str' pointer to a 'u32' pointer, the pointer may be misaligned.
 the pointer type conversion in such way would break the alignment requirement of ptr::read_volatile, and it might lead to unsound behaviors.

**PoC**:
I'll use println instead of print_hex(m) for convenience. But the logic should be same.
```
pub fn print_strmem(s: &str) {
    let p = s.as_ptr();
    let m = unsafe { core::ptr::read_volatile(p as *mut u32) };
    println!("Read value: {}", m);
}

fn main() {
    let s = "abc"; 
    print_strmem(s);
}
```
Result：
```
thread 'main' panicked at library/core/src/panicking.rs:221:5:
unsafe precondition(s) violated: ptr::read_volatile requires that the pointer argument is aligned and non-null
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
thread caused non-unwinding panic. aborting.
Aborted (core dumped)
```

lwz23, 2025-01-06:
I think same problem for https://github.com/oreboot/oreboot/blob/16efa1db9ff220a72dab728b03dbb2dc462b2d76/src/mainboard/starfive/visionfive1/bt0/src/dram_lib.rs#L30
```
pub fn read(reg: usize) -> u32 {
    unsafe { read_volatile(reg as *mut u32) }
}

pub fn write(dest: usize, val: u32) {
    unsafe {
        write_volatile(dest as *mut u32, val);
    }
}
```
they are pub function, but without any varify for the args, they can be invalid parameter and result in UB. According to rust's security specifications, any pub function that can cause UB should be marked as unsafe. I suggest mark them as unsafe or remove the pub field.

orangecms, 2025-01-12:
I've seen your spam in https://github.com/r9os/r9 already.

Read before you post.

The code here in oreboot

- is no-std, i.e., bare metal where we _define_ `print!` (!)
- does not print a string, but a pointer to it (the address) in hex (!)
- is for debugging the very `print!` macros

None of your statements apply.

orangecms, 2025-01-17:
Closing as this is not applicable and there was no further response.

closed 2025-01-17 by orangecms