oreboot/oreboot #771: rustc 1.78.0 is not supported by the following packages

kanakshilledar, 2025-02-27:
Hi,
I am on Ubuntu 24.04. In a fresh installation I am setting up the oreboot build environment and getting the following error during `make firsttime`.

```shell
error: failed to compile `cargo-tarpaulin v0.27.1`, intermediate artifacts can be found at `/tmp/cargo-installIR6o7E`.
To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.

Caused by:
  rustc 1.78.0 is not supported by the following packages:
    litemap@0.7.5 requires rustc 1.81
    zerofrom@0.1.6 requires rustc 1.81
  Try re-running `cargo install` with `--locked`
make: *** [Makefile:36: firsttime] Error 101
```

Do you suggest upgrading the rust compiler version from `1.78.0` to something newer such as `1.84.0`?

Thanks!

kanakshilledar, 2025-02-27:
Full error log: https://pastebin.com/EtyfnvM4

orangecms, 2025-02-27:
Ah, tarpaulin again.

It was set up to report test coverage.
However, since we do like 99% non-testable MMIO for DRAM bringup and have very little testable library code, my pledge is to just get rid of it. That also requires removing its integration.

So let's do it all

1. remove tarpaulin and its integrations
2. bump the Rust version used for the tools (main Makefile)
3. upgrade the other tools (dprint etc)

Want to file a PR for each, in that order? :)

orangecms, 2025-02-28:
Funny, CI reproduced the issue:
https://github.com/oreboot/oreboot/actions/runs/13579498724

We definitely need to fix this now. 🙃🧐

kanakshilledar, 2025-02-28:
Sure would love to file a PR for them :)

I am creating a list of issues to be fixed
- [ ] remove tarpaulin and its integrations
- [ ] bump the rust version used for the tools (main Makefile)
- [ ] upgrade the other tools (dprint etc)

closed 2025-03-06 by orangecms