oreboot/oreboot #772: (de)compression

orangecms, 2025-02-27:
So far, we are using the lzss crate for (de)compression to fit a kernel Image on a SPI flash.

This has its pros and cons, and we can do better, with drawbacks.

lzss takes a lot of time to compress, decompression is mostly fine.

Note that compression happens on the host machine that builds the firmware image, so any cost there is mostly a one-time cost and only "bad" for the development experience when it is slow.

The compressed size is okayish; given a 17 MB Linux Image that already includes an xz-compressed u-root initramfs with many commands (nothing to expect in that part), we end up with 11 MB using lzss.
In comparison, the same Image is compressed much faster and down to 11 MB using zlib-rs and the highest compression level (9).

However, zlib-rs requires an allocator if we want to use it for decompression: https://github.com/trifectatechfoundation/zlib-rs/issues/309
So do https://docs.rs/zopfli/latest/zopfli/ and https://github.com/Frommi/miniz_oxide.

OTOH, lzss works without a heap allocator.

Is there a better (de)compression library that we can use without an allocator? Should we implement one in order to get zlib compression?
It looks like almost all libraries need an allocator: https://blog.logrocket.com/rust-compression-libraries/

dancrossnyc, 2025-03-03:
I was surprised to see that you said `miniz_oxide` requires an allocator, but I remembered that I provide one in `phbl` so perhaps I never noticed. Regardless, it appears that you can turn it off in `Cargo.toml`.  From https://crates.io/crates/miniz_oxide,

>Using the library with default-features = false removes the dependency on alloc and collection crates, making it suitable for systems without an allocator. Running without allocation reduces crate functionality:
>
> * The deflate module is removed completely
> * Some inflate functions which return a Vec are removed

Here, the issue would be that you wouldn't have an `inflate` that returns a `Vec<[u8]>`, but the non-allocating `decompress` function works just fine if you've got a preallocated buffer you can inflate into that's big enough to hold whatever you're decompressing. See e.g., https://github.com/oxidecomputer/phbl/blob/b81e12aff5e9db11b0b0437b3c86352f6c1fff39/src/main.rs#L43

orangecms, 2025-03-04:
Oh excellent, thank you for taking a closer look!

I will try it out, will be easy now with our QEMU port for drafting and the K230 for a real hardware test.

orangecms, 2025-03-05:
Alright, `miniz_oxide` does the job just fine - PR to follow for our QEMU target.

Additional thought: I see no reason to make payload compression optional, given that size is such a string constraint here. Just the integration with the build system will require some work. We already have that for the D1 (sunxi/nezha) anyway.

orangecms, 2025-03-05:
I also looked at the code size changes.

Without compression, the oreboot QEMU port code is 34K.

- with `zlib-rs`, it grows to 87K
- with `miniz_oxide`, it's only 54K.

orangecms, 2025-03-05:
> I was surprised to see that you said `miniz_oxide` requires an allocator, but I remembered that I provide one in `phbl` so perhaps I never noticed. Regardless, it appears that you can turn it off in `Cargo.toml`. From https://crates.io/crates/miniz_oxide, [...]

Turns out that was an inconsistency across the docs. The main README of the repo clearly stated that an allocator is required for no_std, nothing else, so I looked no further.

So I filed https://github.com/Frommi/miniz_oxide/pull/168 to make it consistent.

PR #781 (2025-03-08): switch to miniz_oxide for compression
This required adjusting the sunxi/nezha main stage due to the API change, but also because we need a bigger stack now.
While at it, rework the constants and prints a bit.
With the experience gained now, more rework will follow.

This resolves #772 for a start. We have yet to come to a clear design for building and flashing.
closed 2025-03-08 by orangecms