oreboot/oreboot #776: Unsoundness issue in

lwz23, 2025-03-03:
Hello, thank you for your contribution in this project, I an testing our static analysis tool in github's Rust project and I notice the following code:
https://github.com/oreboot/oreboot/blob/1dfdc9e1dfb05ad5e158874724b54655346e75b1/src/mainboard/sunxi/nezha/bt0/src/mctl.rs#L216
```
fn readl(reg: usize) -> u32 {
    unsafe { read_volatile(reg as *mut u32) }
}
```
I think there is a unsound problem because this function doesn't varify the `reg` is valid and pass it to unsafe function `read_volatile`. It will trigger UB. Although it is a private function, I notice a possible way to call this function from a pub function `init_dram`.

pub fn init_dram -> fn readl

```
/// # Safety
///
/// No warranty. Use at own risk. Be lucky to get values from vendor.
pub fn init_dram(para: &mut dram_parameters) -> usize {
    if para.dram_tpr13 & (1 << 16) > 0 {
        if VERBOSE {
            println!("DRAM only has internal ZQ.");
        }
        writel(RES_CAL_CTRL_REG, readl(RES_CAL_CTRL_REG) | 0x100);
        writel(RES240_CTRL_REG, 0);
        sdelay(10);
    } else {
        writel(ANALOG_SYS_PWROFF_GATING_REG, 0);
        writel(RES_CAL_CTRL_REG, readl(RES_CAL_CTRL_REG) & !0x003);
        sdelay(10);
        writel(RES_CAL_CTRL_REG, readl(RES_CAL_CTRL_REG) & !0x108);
        sdelay(10);
        writel(RES_CAL_CTRL_REG, readl(RES_CAL_CTRL_REG) | 0x001);
        sdelay(20);
        if VERBOSE {
            let zq_val = readl(ZQ_VALUE);
            println!("ZQ: {}", zq_val);
        }
    }
```
I can even see the `# Safety No warranty. Use at own risk. Be lucky to get values from vendor.` It shows you know it may not as safe as it decleared. However, I want to point that according to rust's safety guarantees, any function that could cause undefined behavior should be marked as unsafe, rather than marked as safe and then annotated as `No warranty`, as is currently the case. I think a appriciate solution would be mark it as unsafe.

orangecms, 2025-03-03:
We had this discussion here before, see also #761.

Keep the context of this project in mind.
This is firmware, and no documentation on those parts exists.
99% is MMIO, and the implementation is based on the vendor code.

orangecms, 2025-03-03:
This is the third time I'm reading from you and not expecting a response.
I am closing the issue accordingly. Feel free to reply again.

lwz23, 2025-03-03:
Thank you for your reply. I understand that as firmware code, this project has its specific context and challenges, especially working with MMIO in an environment with limited documentation.

I raised this issue purely from the perspective of testing our static analysis tool and considering Rust's safety guarantees. I see that similar issues have been discussed previously in #761, and I understand the project team has an established position on this matter.

As you mentioned, this approach may be acceptable within the context of your project, and I respect your decision.

Thank you for taking the time to respond, and I wish you success with the project.