oreboot/oreboot #784: Are there any plans to support x86 boards in the future?

sameershaik, 2025-03-23:
I would like to inquire if there is a current maintainer for the x86 boards and whether support for these boards will be reinstated in the future. Additionally, would you be open to merging the FSP and ME binaries back into the upstream code?

orangecms, 2025-03-23:
Hi, we already had some initial attempts; see the commit history and graveyard.md. Due to lack of documentation, we shifted our focus to other platforms.

Feel free to contribute, we'll gladly assist if you have any questions. :)

orangecms, 2025-03-23:
Regarding maintenance and binaries:

Rather than wrapping blobs with some parameters and logging, our goal is to have actually open implementations. However, we know that some chip vendors will not cooperate in that regard. If you happen to have documentation on vendor blobs and would like to work with them, we won't stop you.

There is noone at this point driving such efforts though, so that will be up to you. Note that ME binaries are usually not provided with a license allowing us to redistribute them. The only exception I have seem so far was a single binary for some server platform that was once posted to the EDK2 mailing list, IIRC.

orangecms, 2026-02-15:
Should be clear, closing now after about a year of inactivity.